ludiazv/node-nrf24 #1: ReadMe seems to be out of date

tobiasmuecksch, 2018-03-18:
Hi @ludiazv ,

thanks for this awesome library. I'm currently trying to use it, but the ReadMe seems to be out of date.

_For example:_

- Instead of `var nrf24 = require("nRF24");` you have to `const nrf24 = require("nrf24");`.
- `var rf24 = nrf24.RF24(22, 0);` does not work, you have to instantiate it instead: `var rf24 = new nrf24.nRF24(22, 0);`

If pull requests are welcome, I will create one - in case I find out on my own how to run this library.

ludiazv, 2018-03-19:
Thanks. That's True. To create the radio object **'new'** keyword is mandatory. In fact the library raises an exception informing about that.

 The documentation need improvement and correction. I need to review it as I have also spotted some errors on it. That need correction but yet haven't found time to do it.

I'm waiting for an update in the C++ library to update the library. In principle it should support:

- Review of documentation: Still not complete and with errors as you have identify.
- Document RF24Mesh & additional testing.
- First Implementation of RF24Gateway directly on nodejs.
- Better exception management.
- IRQ line support to improve thro

PRs and Testing feedback are welcome as this node package is very new.  What board ,OS, nodejs version are you using?

Regards,

closed 2018-03-23 by ludiazv

tobiasmuecksch, 2018-03-23:
Sorry for the late reply... I'm using a raspberry pi 3B and a raspberry pi zero. Both on the latest raspbian (Releasedate 2018-03-13)

ludiazv, 2018-03-23:
I updated some days modifications on the documentacion and improved some points of the code. It should be working on your setup. As I have tested in a Zero W without problems.  Are you still having problems?

Regards,

tobiasmuecksch, 2018-03-23:
I got it working perfectly. I just wanted to let you know, that the information in the readme isn't correct at some spots. 
But still, this is an amazing library. Thank you very much for that - you saved me hours.

Next week I will have some spare time, and will evaluate where I can help with some improvements.