ludiazv/node-nrf24 #2: How to send big buffers (4000 bytes)

tobiasmuecksch, 2018-04-03:
Hi @ludiazv,

I'm having a problem. I need to transmit buffers with a size of 4000 bytes. This package can't transmit more than 32 bytes.
Do you have any idea or hint how to transmit bigger buffers?

ludiazv, 2018-04-03:
32 bytes is hardware limit for a single frame. There is no way to send more than that in a single frame. This is typically called MTU in network protocols. For this radios the MTU is 32.

For sending 4K you need to implement fragmentation of the data in 32bytes chunks/fragments. A simple way could be to add a 2-byte header in your payload with the packet number & data length.

On transmission, split the original buffer in 30 byte chunks, pack each pakcket with the header and write each packet to the radio. You shold implement controlled retry sending each chunk

On the receiver, defragment the sequence controlling the order and the size of each chunk.

tobiasmuecksch, 2018-04-04:
@ludiazv Thank you very much for the detailed in-depth answer. 

**FYI:**
I'm currently trying to make an audio transmitter that transmits the microphone signal from a raspberry pi to another raspberry pi. Currently I'm trying to transmit the raw signal, but maybe that won't work.

The maximum speed of this module is 2 Mbps = 250 kB/s am I right? That means 2500 / 32 Byte = ~78 Frames per Second. The raw signal currently is ten times this big.

Conclusion; I have to compress the signal.

closed 2018-04-04 by tobiasmuecksch