ludiazv/node-nrf24 #3: Packages arrive only when auto_ack: true

tobiasmuecksch, 2018-04-04:
I'm having a sender and a receiver.

The sender constantly sends packages, and the receiver prints out the number of packages received per second. 

When I disable auto_ack, **less than 1%** of the packages arrive.
Only when auto_ack is enabled, all the packages arrive.

I assumed that at least 50% of the packages will arrive. But less than 1% seem strange to me.
Is that normal?

ludiazv, 2018-04-04:
Hi,
  This seems like a error. Could you share your code to do some test?

Regards,

ludiazv, 2018-04-04:
I have tested your code and found on problem on the library. The writingPipe is always configured for AutoAck. I did some modifications so now you set up the writting pipe with no Auto Ack.

I've have also make this optional parameters. I will do some addtional testing and commit the changes.
This should solve the issue.

ludiazv, 2018-04-09:
I've tested this bug and works well. It should be ready in the next release.

Regards,

tobiasmuecksch, 2018-04-10:
@ludiazv Thank you so much, for your awesome work!

ludiazv, 2018-04-18:
@tobiasmuecksch I just uploaded a new version to the master branch. It should work now with and without Ack. The API has improved. But it has breaking changes on ``read`` method. Now it receives an array.

Could check if you are able to send and receive without ACK.

closed 2018-06-25 by ludiazv