ludiazv/node-nrf24 #5: Are there future plans to support the protocol used in RF24Network or RF24Mesh?

ristomatti, 2018-04-04:
@ludiazv I noticed you mentioned nRF24/RF24Audio on #2 discussion. I've been using/testing the RF24Mesh library by the same developer/team on my Arduino/ESP8266/ESP32 microcontroller projects. Are there plans to make node-nrf24 to be able to interact with devices using those libraries?

I've been looking into ways to make Raspberry Pi's to communicate using the same radios and this is the main reason your library caught my attention. RF24Mesh is also supported on the Raspberry Pi but although I've started getting the hang of C/C++ for Arduino, I'm still a long way understanding how to make use of the same on a Linux environment. On the other hand I have a solid JavaScript background so being able to use Node instead would be awesome.

ludiazv, 2018-04-04:
RF24Network is not planned (I do not see the advantages of having this layer having the mesh option) but RF24Mesh is implemented but not documented yet.
In fact I'm using mesh with node-red using the sister project https://github.com/ludiazv/node-red-contrib-nrf24.

The use of mesh is very simple and similar to orginal libraries. In fact in test folder you can find a test js for mesh master and arduino scketch working on atmegas and ESP8266. I didn't test with ESP32 yet.

Regards,

ludiazv, 2018-04-06:
Thanx,
  If are going to test Mesh in NODE-RED please let me know if you found any issue.  But if possible open it in the other repository.

Any feedback is wellcome as the mesh is less tested from my side. I use in master mode with low speed (250kbps) receibing information from 3 mcus with sensor information an sending commands to switches. The finformation flows from the MESH <-> MQTT.

 It works fine but information exchange happens with very low frequency. I have not tested it with a lot of nodes or contious exchange of information at very high frequency. So the overheads and response times of Node-RED are not testes with high load.

ristomatti, 2018-04-06:
@ludiazv Sure! Though I'm currently writing a bidirectional RF24Mesh<->MQTT gateway library for ESP8266 which is/was my original intention on solving the problem of connecting nodes w/o wifi/out of reach from wifi. I'm currently running Node-RED on a network 200km away from the nodes I want to receive data from so MQTT would be the most natural fit. But if this fails miserably or I want to build a separate "autonomous" Node-RED system running there, I will definately give it a try! :)

In case you're interested in my project, I will publish it to GitHub if/when I'm at a point it could be useful.

ludiazv, 2018-04-07:
Nice Project. The ESP solution will be cheaper than using a SBC such a PI of course. But SBC will bring you more flexibility and the development will require little coding only in JS. It would easy to maintain as you can avoid OTA updates of the ESP device by simply login to the remote gw using ssh or node-red console.

 If you have a Ethernet availble in the gw location you can use an Orange Pi Zero mini node-red node that act as gw mesh<->mqtt for 12 USD more or less. A low cost 8GB SD will do and the 256MB verions of this board is arround $7. In this node you could even install a local mosquitto broker so the latency between the mesh and mqtt will be near to zero. The local broker can be brideged with your main broker for the desired topics (even with name remaping wich will bring addtional flexibility in the future). You can avoid in this way using wifi and ardunio libraries for upstream management which has in some cases limited QoS compatibility and other limitation such package size,... by using battle-tested transports such ethernet and mosquitto. In addition, you can also can implement a local data logger, vpn, ... using standard linux/node-red tools without any additional coding.

The radio and mesh part in this library is not mature however so you should test this well before going on with this approach. Needless to say that it will help to debug and document current mesh implementation. If you do some progress here please let me know.

Regards,

ristomatti, 2018-04-08:
Yes I have been thinking about something very similar as an alternative. I even have a RPi3 there on the site used as a camera and acting as reverse proxy for couple of network cameras.

The main reason why I decided to use the ESP8266 is that there are quite frequent power outages and I've found RPi's quite unreliable to boot up after sudden loss of power. UPS solutions seem to be either relatively expensive/difficult to build yourself and to be frankly overkill for the purpose in this case. The ESP8266 might not be too reliable but the great thing with it like any other microcontroller is that you can for example install a physical timer to toggle it's power once in a day.

The second reason was my inexperience with C/C++ on Linux so that I could've used the RF24Mesh library. Your library could now possibly change this as writing a similar MQTT gateway in Node.js would be trivial. Although I have to say I've found working with external devices using the RPi quite a pain compared to using microcontrollers w/ abundant available Arduino libraries. It's just not so straightforward as often need to manually build & install libraries and their dependencies, run the code as root etc.

Regardless of the way I'm going to implement the needed communication I might be curious enough to try your library anyhow. I'll be reporting back if I find some issues when that day comes.

ludiazv, 2018-04-09:
If you need a UPS as you say Rpi solutions are to expensive due to the need of supporting high currents and boost converters to 5V or expensive battery packs. ESP will be much better at this point as you can use a 3.7V Lipo Cell with a cheap charger module and a good LDO. This will give you a lot hours of operation without power. Also if your GW needs to interact with other perfifericals mcu is much better. ESP will be very limited as SPI interface will "eat" most of the io lines of the mcu. Dont forget to use modem sleep in my experience this mcu is not low power friendly. 

The esp has a decent watchdog that could reset you gw if something goes bad. I think you are not going to need the external reset.

If you have a Rpi in place still is a good idea to put a Local MQTT broker running there. You could reduce the latency for ESP so it can quick transfer and rcv messages from the broker having more time to deal with the perifericals such the radio. ESP has only one core shared and all wifi communications are managed by core. if they take to long you code will not be executed when needed by the radio. Also use static IP if possible.