ludiazv/node-nrf24 #6: RF24 HARDWARE FAIL: after first write

comac2k, 2018-05-17:
First write works fine and is received at the other end. But any write after that blocks forever in an infinite loop of "RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc." error messages.

This happens with my own test program in js, with test/test_RF24.js (when invoked with parameter "1") and also when used from NodeRed.

The C++ test program from the RF24 library works fine using the exact same version of the C++ lib running in the same hardware. And so does the C++ version of my test program. They can send as many transfers as needed with no error.

ludiazv, 2018-05-21:
Did you clone the repository or is NPM install?

comac2k, 2018-05-21:
By the time I created the issue I had only tried "npm install". Checking out master with git results in ALL packets failing with the "RD24 HARDWARE FAIL" message. Even the first one (which works fine and I can see it's received on the other end when using "npm install").

Another difference is that git version seems to print additional debugging info after configuring the interface.

ludiazv, 2018-05-21:
Yes, the master has test traces for debugging on config. The behavior you describe is very strange as this error is typically issued with HW and wiring problems. But if a cpp program using the library works the node module should work also as it simply call library.

Could you better describe your set-up, test code and output?

I can try to replicate your problem in my RPi3. I use dietPi though but I think this is not the soruce of the problem.

Regards,

comac2k, 2018-05-21:
I've tried checking out several commits from git in an attempt to isolate the issue, but all I could come up with is this:

commits from 8635de7b4deb0034d9949923dc7467001fc21460 to 220d7661776aa4694ac3ecfaac195da17332e3ec -> fail to build
commits from 4077e5f050081a9d7db5555692e3578a5187ae00 to 72a226bef49e07734b4d3af0d4a8a748b1ae1375 -> first transmission works, all others fail
commits from d9c741e0da06d2a52d2fa2db100add72292a3ffc to bcd6f1b9dc2c45e57ebacfd88872a7afdbba5f6d -> fail to build
commit 33ddea3ce0ca9e6f21a3c58e382524ad670bb419 -> all transmissions fail

comac2k, 2018-05-21:
About my setup:

Raspberry Pi 3 model b.
Running Raspbian GNU/Linux buster/sid
nodejs version: v8.11.2
kernel: 4.4.38-v7+

I'm using a nRF24L01+ module with embedded antenna both in this raspberry and in the arduino receiver. The only difference with the pinout in your readme is that I'm using pin 25 for CE (instead of 22).

My test program reads lines of text and sends them. These lines contain only 4 characters that can be either 0 or 1. Upon receipt of a message, the arduino will enable / disable 4 relays based in those 4 characters.

Here's the code for reference:

```
var nrf24=require("nrf24");
var readline = require('linebyline');

var rf24= new nrf24.nRF24(25, 0);
rf24.begin(true);
rf24.config({
  PALevel: nrf24.RF24_PA_MAX,
  DataRate: nrf24.RF24_1MBPS
});
rf24.useWritePipe("0x544e4c4331");

var rl = readline(process.stdin);

rl.on('line', function(line){
        success = rf24.write(Buffer.from(line));
        console.log("Sent " + ( success ? "OK" : "KO" ));
});
```

I tried using node-red node for rf24, based on this library, and I get the same problem: first message works, further messages enter an infinite loop of errors, freezing the node-red instance until I kill the process.

Is there anything else I could do to help debug this?

ludiazv, 2018-05-21:
I will try to use your code to test it.  To have addtional trace on the init of the radio you can:
``
rf24.begin(true)-> This will display the status of radio after initialization.
rf24.config(config,true) -> This will display the status of the radio after configuration.
`` 

- What is the output of the exectution?
- Does line > 32 chars?

On the other hand: Did you compile and install the RF24 Library with the script in this repo? Have you reboot after installing the lib in SPIDEV mode?

The library for node-red will fail if this fails. So is better to isolate the problem the use it in node-red.

comac2k, 2018-05-21:
And just for reference, this is the C++ version of the program and it's output.

The program:
```
#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include <unistd.h>
#include <RF24/RF24.h>

using namespace std;

RF24 radio(25,0);

const uint8_t pipeId[] = "1CLNT";

int main(int argc, char** argv){

  radio.begin();
  //radio.printDetails();

  string input = "";

  radio.openWritingPipe(pipeId);
  radio.printDetails();
  radio.stopListening();

  while (1) {
    getline(cin,input);

    bool ok = radio.write( input.c_str(), 4 );
    printf("Sent %s.\n", ok ? "OK" : "KO");
  }

  return 0;
}
```

Output:
```
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x544e4c4331 0x65646f4e32
RX_ADDR_P2-5     = 0x33 0xc4 0xc5 0xc6
TX_ADDR          = 0x544e4c4331
RX_PW_P0-6       = 0x20 0x20 0x20 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x07
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
1000
Sent OK.
0100
Sent OK.
0000
Sent OK.
^C
```

ludiazv, 2018-05-21:
thx, I will try to reproduce this.

ludiazv, 2018-05-21:
Hi,
 I just tested your code in my RPI. It works fine with the current version of the repository. I was NOT able to replicate your error. 

I'm using very similar configuration:
Rpi3 model B.
Node version: v8.11.1
Linux: 4.14.30-v7+ DietPI
Driver:SPIDEV with similar access rights for /de/spidev & gpios

I can't figure why it fails on your setup. The main difference of the cpp code and the js code is that that in the library perform several actions during begin & configure (in fact it mask irq and also startlisteing for  short period) but for transmitting it uses a simple Write call to the library. Nothing special.

Read this https://github.com/nRF24/RF24/issues/205 about similar problems with the library.  The problem you are having is due to timeout waiting for the status after write to the radio. The radio is not responding as expected.

You can try several things:

- Check your power of the radio. I made hat with capacitors and dedicated power supply for the radios. Check it here: https://github.com/ludiazv/borosRF2 for the schematic. 3V3 pins from the PI should work but users have reported problems, specially on clones. Also check that you have beefy decoupling capacitor in the radio pins VCC-GND.
- Run your sample code after running the cpp version but without calling configure.
- Unplug and plug the radio to reset it.
- Check your wires again and make them shorter if possible.
- Try to run cpu-freq at constant speed without throttle to avoid changes on SPI speed. (this a RPI problem affecting I2C & SPI).
- Try to use debian strech - stable instead of buster/sid (testing).
- Try to compile the library in RPI mode. (you will require sudo to run the program).

comac2k, 2018-05-22:
Thanks!

That post was very helpful when I had wiring issues (I connected the CE and CSS pins backwards initially), but that's something I fixed before I wrote my own test program.

Hardware modifications to fix a software issue are not acceptable (besides, adding a capacitor between GND-VCC results in lots of packets lost in my experience). Nor is changing the operating system.

I'll have to either fix it myself or write my own wrapper around the C++ library. Whatever turns out to be faster to do.

ludiazv, 2018-05-22:
Ok. If you find the source of the problem PR are welcome. I suspect that the key would be in the config method as it try to reset the radio.maybe commenting the all the lines but stopListening in _resetState() method inside rf24_setup.cpp. 

Cap is not a HW solution for a software problem. Capacitor are not needed if your supply is good at reacting to variable loads. With NRF24L01 radios currents spikes are typical during transmission of each packet if the power line can't cope with this spikes the radio can fail (this spikes are worst in clones). This happens also when the radio changes from tx to rx. In principle rpi3.3v pins should be sufficient but there are reports where not. So capacitors needed to solve a hw problem. Software can have as they can treat stress the hw.

ludiazv, 2018-09-26:
This an old issue but just a question. 
Was your OS 64 bits or 32?