ludiazv/node-nrf24 #7: print_details not working for config() function

ngfelixl, 2018-05-27:
Hey,
this seems to be a great library, thanks! While the lib successfully and correctly prints true for `.begin()`, `.present()` and `.isP()`, calling `config` with the print_details flag enabled like follows

```javascript
radio.config({
  PALevel: nrf24.RF24_PA_MIN,
  [...]
}, true);
```
The code in a try catch environment prints the following error with the line number pointing to the config function
```
SyntaxError: Unknown error
```
Any suggestions?

ngfelixl, 2018-05-27:
Also the GettingStartet.js works only if I use `config(config)` without print_details. And breaks at line 41 (sender) or 27 (receiver) at
```
radio.addReadPipe(Pipes[0]);
```
The related error is
```
readline.js:1021 < throw err;

SyntaxError: Unknown error
  at snd (/home/pi/GettingsStarted.js:41:9)                           # sender @41:9, receiver@27:19
  at rl.question (/home/pi/GettingsStarted.js:83:31)
```

ludiazv, 2018-05-27:
I think your are using the previous version that is on npm repository. Use master branch to use this API call.

Regards,