ludiazv/node-nrf24 #9: RX pipes

acegla, 2018-11-23:
Hi, thank you for great job with integrate NRF24 with nodejs.

Unfortunately I have problem with identify pipe on read.
In your docs on read arg `.pipe `should be the same what register by `.addReadPipe`.
Now in simple code 
  ```
var rpipe = radio.addReadPipe(Pipes[0]);

  console.log("Read Pipe " + rpipe + " opened for " + Pipes[0]);

  radio.read(
    function(d, items) {
      for (var i = 0; i < items; i++) {
        console.log(`in ${d[i].pipe} rpipe ${rpipe}`);
}})
```
I've only receive 
```
Read Pipe 1 opened for 0xaaaaaaabbc
in 2 rpipe 1
```
Doesn't matter what I change before `.read` I've always reveive `.pipe `=2

Do you have any idea or explanation why ?

ludiazv, 2018-11-28:
Hi
 This is no intended. Are you using master branch on the repo?
  I will test this later this week.

Regards,

ludiazv, 2019-05-19:
Reviewing past issues on the lib while including some changes I tested this with no problems.
i will close the issue. Please reopen if needed.

closed 2019-05-19 by ludiazv