ludiazv/node-nrf24 #10: Unable to connect radio

bjrenfrow, 2019-01-30:
I'm using the current master branch.
And running the code with a PI B+.
CE => GPIO25
CSN => SPICS0

```
const lib = require('nrf24');

const radio = new lib.nRF24(25, 0);

radio.begin(true);

radio.config({
  PALevel: lib.RF24_PA_LOW,
  DataRate: lib.RF24_1MBPS,
  Channel: 76,
}, true);

console.log("Radio connected:" + radio.present());
console.log("Is + Variant:" + radio.isP())

const readPipe = radio.addReadPipe('0xf0f0f0f0e1', true);

const pipeNum = radio.read(
  (data, n) => {
    console.log('read', data, n);
  },
  (stopped, user, errorCount) => {
   console.log('stopped', stopped, user, errorCount);
  }
);

console.log('pipeNum', pipeNum);
console.log('state', radio.getStats());

process.on('SIGINT', () => {
  radio.stopRead();
  radio.destroy();
  console.log('Bye, bye!');
  process.exit();
});
```
Results:
```
pi@raspberrypi:~/dev/radio $ sudo node radio.js 
Radio details after config:
===========================
STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	 = 0x0000000000 0x0000000000
RX_ADDR_P2-5	 = 0x00 0x00 0x00 0x00
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x00
RF_CH		 = 0x00
RF_SETUP	 = 0x00
CONFIG		 = 0x00
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MIN
IRQ gpio (negative if disabled):-1(0)
Config internals:
check the values as incorrect values fallback to default values.
 PALevel:1 DataRate:0 Channel:76 CRCLength:2 Retries:(5+1*250us,15 attempts) PayloadSize:32 AddressWidth:5 AutoAck:1 TxDelay:250 PollBaseTime:35000
===========================
Radio connected:false
Is + Variant:false
pipeNum undefined
state { TotalRx: 0,
  TotalTx_Ok: 0,
  TotalTx_Err: 0,
  PipesRx: [ 0, 0, 0, 0, 0, 0 ] }
^CBye, bye!
```
Am I doing something wrong? I'm not able to connect the radio or get a pipeNum?

ludiazv, 2019-01-30:
I did not test on b+. But in principle it should work.  By the trace you show seems is a wiring problem. Have you test a plain cpp version?

bjrenfrow, 2019-01-30:
I've checked the pins again: Here's my setup.
[image]
[image]
[image]
[image]

I haven't tested the cpp version. The docs are terrible. What do you recommend?
Thank you, and sorry if this is irritating.

bjrenfrow, 2019-01-30:
OMG... am I missing this: https://www.amazon.com/gp/product/B01IK78PQA ?

ristomatti, 2019-01-30:
@bjrenfrow Yes, unfortunately you only have the adapter that is required if NRF24L01 modules are used with microcontrollers that use 5V logic (like Arduino Uno or Arduino Mega). The modules expect 3.3V logic signals, that a Raspberry Pi uses. There are a few different types of NRF24L01 radios available. There are the ones you linked, then smaller ones with a lower range that are probably the most common which look like this:

[image]

There's also a shielded version of the longer range one you linked that looks like this (antenna is not shown in the picture but it typically comes with a similar longer antenna):

[image]

If you need the modules to communicate over longer distances, I would recommend the above one. It is the most expensive one though but from my experience it is much easier to work with. The unshielded modules are more sensitive to external factors like other nearby components or electric devices.

bjrenfrow, 2019-01-30:
Thank you so much @ristomatti ! My eyes are opened. I can't believe I was trying to get signals from a piece of plastic.

closed 2019-01-30 by bjrenfrow

ludiazv, 2019-01-30:
Just one comment, if you are going to use the power adaptor connect VIN with 5V of your PI. The power power adaptor will regulate the voltage down 3.3V.

Without the power adaptor using the 3.3V power of the PI would work but possibly you need to solder some cap on the radio as the power supply of the PI have current and line regulation limitations. Using PA/LNA radios (longer range as you show) will not work stable as this radios require considerable more power.

If you are looking for more permanent solution check the phat I designed. It solves not only wiring but also power issues with dedicated beefe capacitor. Also supports dual radio configuration. Check it here: https://github.com/ludiazv/borosRF2

bjrenfrow, 2019-01-31:
This is slick. I most likely will want to start producing lots of these, and having a hat I can just plugin will be essential to me. Thanks @ludiazv!

ludiazv, 2019-01-31:
@bjrenfrow I still have some boards for a small batch i made. If you are insterested let me know.

Regards,

bjrenfrow, 2019-02-08:
Please let me know if I'm abusing this issue channel. I'm stuck on one more thing, if I can get this to work I definitely will be purchasing some units.

I got the arduino to send to the raspberry pi! I want to send something back to the arduino now. The following happens with this code:

PI:
```
const lib = require('nrf24');
const radio = new lib.nRF24(22, 0);

radio.begin(true);
radio.config({
  PALevel: lib.RF24_PA_MAX,
  DataRate: lib.RF24_1MBPS,
  Channel: 76,
  CRCLength: lib.RF24_CRC_16,
}, true);

console.log("Radio connected:" + radio.present());

radio.useWritePipe("0x0F0F0F0F0F");

setInterval(() => {
  radio.write(
    Buffer.from('Hello'),
    (...args) => console.log(args)
  );
}, 5000);
```

Arduino:
```
//Send.ino
#include<SPI.h>
#include<RF24.h>

// ce, csn pins
RF24 radio(9, 10);

const uint64_t initialReadingPipe = 0x0F0F0F0F0FLL;

void setup(void){
  Serial.begin(115200);
  delay(1000);
  radio.begin();
  radio.setCRCLength(RF24_CRC_16);
  radio.setPALevel(RF24_PA_MAX);
  radio.setChannel(76);
  radio.setDataRate(RF24_1MBPS);
  radio.setAutoAck(true);
  radio.openReadingPipe(1, initialReadingPipe);
  radio.powerUp();
  radio.startListening();
  Serial.println("Start...");
}

void loop(void){
  if (radio.available()){
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.println(text);
  }
}
```

The PI outputs:
```
===========================
Radio details after config:
===========================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xc2c2c2c2c2
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x7e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
IRQ gpio (negative if disabled):-1(0)
Config internals:
check the values as incorrect values fallback to default values.
 PALevel:3 DataRate:0 Channel:76 CRCLength:2 Retries:(5+1*250us,15 attempts) PayloadSize:32 AddressWidth:5 AutoAck:1 TxDelay:250 PollBaseTime:35000
===========================
Radio connected:true
[ false, 0, 0, 1, 32, false ]
[ false, 0, 0, 1, 32, false ]
[ false, 0, 0, 1, 32, false ]
[ false, 0, 0, 1, 32, false ]
```
The Arduino only outputs
```
start...
```
Because it sent to the pi, I'm sure everything is wired correctly. Any ideas why I cannot send back to the arduino. Thanks!

bjrenfrow, 2019-02-08:
@ludiazv how do you mount the borosRF2 onto the pi, it looks like its sitting on top of a special case. Do you have any recommendations?

ludiazv, 2019-02-09:
Hi @bjrenfrow ,
  Your code is ok (assuming that you are using current master branch) I tested on my rig and works fine. The only difference is that I use a different pinout for arduinio (7,8 pins for ce & cs) and gpio 24 for the Pi.  I used a nano & rpi3b (not b+). So I don't think is sw related.

As for your description, I understood that you can send frames from arduino -> Pi but not from Pi -> arduino. TX problems are usually related with insufficient/poor filtered power to the radio. Specially on PA/LNA radios. They require a lot power to transmit for a brief period. Are your correctly rcv ACKs from the Pi when sending from the ardunino->pi?

Try some of this:

On the hw side:

1. Check again the wiring and try to make shorter the wires specially VCC & GND.
2. Put a big Cap, say 100uf-1000uF close to VCC&GND of the radio:
[image]

3. Use other PSU for the radio and do not power the radio from 3.3V rail of the PI (use the adaptors with 5V power).
4. Interchange the radio modules or test with new ones. I think is better to start with cheap clones with on-board antenna.
5. To be sure that your modules are ok. test with 2 arduinos using the gettingstarted example of the library.

on SW side:
1. reduce TX power to PA_MIN when testing. if they are close it wont be a problem.
2. Reduce payload size.
3. Reduce transfer speed to 250KBPS

As regards, the case you need a case with high clearance, or use stackable board based on stand offs and acrylic covers. This case you can adapt the height of the box to your needs. I personally don't use any commercial solution as I normally print a specific case that will fit to the project needs.
One caveat: AVOID metal cases they block radio signals. if metal case is a must use a external antennas.

Regards,

bjrenfrow, 2019-02-10:
You are a genus! The fix was to #3. You should write a book. I have two communication now!

If I want to just plug in your module into my bare pi, will these settings work:

[image]

ludiazv, 2019-02-10:
In the second box chose "soldered pin headers".

Regards,

bjrenfrow, 2019-02-11:
Just ordered 5. I will be ordering a lot more if our project is successful. Thanks again for your help.