ludiazv/node-nrf24 #12: Mesh not working with RF24Mesh 1.0.6 but slightly on latest 1.0.7

systemofapwne, 2019-06-10:
I have tried to narrow down, why I'm completely unaible to connect my Arduinos to my RPi while the arduinos (when one of them is a master) can connect to each other in a mesh.

*The symptoms:* 
* If I'm running the mesh on my RPi in master-mode but no arduino-node can connect to the RPi.
* If I have an additional arduino as a master in parallel, all my other arduinos will connect to the arduino-master *and* the RPi master sees the data communication.

*Configuration*
RF24: 1.3.2
RF24Network: 1.0.9
RF24Mesh: 1.0.6 on RPi, 1.0.7 on Arduino

*Conclusion:*
Data communication works, but the DHCP seems to fail, if RF24Mesh 1.0.6.
I updated RF24Mesh to 1.0.7 on the RPI and now, at least, the RPi seems to see the external nodes now. But the external nodes won't accept/see their associated addresses (yet). I will try to further narrow this down.
Note: It also seems to fail, if I downgrade the arduino RF24Mesh to 1.0.6 (so RPi and Arduino share the same lib).

*Suggested fix*:
Tell build_rf24libs.sh to build against RF24Mesh 1.0.7.

systemofapwne, 2019-06-10:
Disregard my statement regarding that DHCP still fails. It turned out, that the nrf24l01 bypass-capacitor on my prototype board fell off.
I can now connect my arduino to the RPi. Changing the library to 1.0.7 on the RPi was the only necessary change.

ludiazv, 2019-06-10:
Thanks for the feeback.  I will bump the version to 1.0.7 as you suggest.

I think the mesh part of the library require some refactoring for perfomance. I hope to find some time to do it this summer. :S

systemofapwne, 2019-06-10:
You are welcome. I'm looking forward for upcoming versions of this awesome lib!

ludiazv, 2019-06-12:
Updated in new release 0.1.3.
PRs are open in case you want to contribute in the future.

closed 2019-06-12 by ludiazv