ludiazv/node-nrf24 #15: Option to set SPI CLK frequency

systemofapwne, 2019-07-08:
Right now, this lib uses the max SPI CLK (set to 10 MHz). This can cause some trouble, especially if cables are too long or if interferences happen. In principle, a proper PCB design or shielded wiring scheme might solve this. Yet, it is not always possible to achieve this goal, especially for prototypes. 
This can cause errors like the radio not responding at all (10 cm long cables will definitly fail) or having it occasionally failing (RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.). 
Allowing one to reduce the CLK speed via the node interface to a customized and lower speed, this problem can be minimized.
In the original RF.24 library, this is done at RF24.begin(). Here, it would suit best at the constructor though: nRF24(CE.CSN,speed = 10MHz).

ludiazv, 2019-07-09:
Hi,
 Yes this would be a nice option. the problem is that the base C++ library do not support a standard way to change this. At compile time there are some options or using specific drivers (Rpi) but not in SPIDEV that is the preferred way as it would be compatible with  SBCs and do not require root grants. I contributed to the c++ library while ago improving the SPIDEV to support that. but in the main code of RF24 class do not support it now.
Check this out: https://github.com/nRF24/RF24/blob/bf259002b2eb0b44df812d7c2d332f70b50525b4/RF24.cpp#L85

If you have any idea please let me know. But to make it possible it would require change de base library.

Regards,

ludiazv, 2020-09-25:
This feature is now available in the base library. And has been ported to the module. To enable it you need only to call the constructor with a 3rd parameter with the SPI frequency in Hz.
The change is now in master branch and will be published in the next version.

closed 2020-09-25 by ludiazv