ludiazv/node-nrf24 #16: [not a bug or issue]please help about  raspi gpio pins and pipe address

Nathanlee2010, 2019-09-19:
```
var rf24= new nrf24.nRF24(<CE gpio>,<CS gpio>);
```

* my raspi  is  module 3b +
* i link  nrf24 and raspi follow this:
```
http://tmrh20.github.io/RF24/
![image](https://user-images.githubusercontent.com/20190746/65207550-8917e300-dac4-11e9-8005-cdcd8f1a8ce6.png)
```
#### question 1 : in my project ，new nrf24.nRF24 should be:
```
var rf24= new nrf24.nRF24(22,8);   
or
var rf24= new nrf24.nRF24(15,24);
??????
```

#### question 2: const byte address[6] = '00001'   How to convert to  "0x0000000000" ?
in c++ （arduino IDE），the read address is "byte address[6] = '00001'"
in nodejs  the write address is ???     

i want that  raspi(nodejs)   send message to arduino(c++)

Please forgive me for taking the liberty and ignorance
I used nrf24 for the first time, Im a javascript developer and i don't understand C++ and hardware development.

Nathanlee2010, 2019-09-19:
##   i link nrf24 and raspi follow this:

PIN | NRF24L01 | Arduino UNO | ATtiny25/45/85 [0] | ATtiny44/84 [1] | LittleWire [2] | RPI | RPi -P1 Connector
-- | -- | -- | -- | -- | -- | -- | --
1 | GND | GND | pin 4 | pin 14 | GND | rpi-gnd | (25)
2 | VCC | 3.3V | pin 8 | pin 1 | regulator 3.3V required | rpi-3v3 | (17)
3 | CE | digIO 7 | pin 2 | pin 12 | pin to 3.3V | rpi-gpio22 | (15)
4 | CSN | digIO 8 | pin 3 | pin 11 | RESET | rpi-gpio8 | (24)
5 | SCK | digIO 13 | pin 7 | pin 9 | SCK | rpi-sckl | (23)
6 | MOSI | digIO 11 | pin 6 | pin 7 | MOSI | rpi-mosi | (19)
7 | MISO | digIO 12 | pin 5 | pin 8 | MISO | rpi-miso | (21)
8 | IRQ | - | - | - | - | - | -

ludiazv, 2019-09-19:
Hi,
 Check RPI pintout in this project documentation:
  https://github.com/ludiazv/borosRF2
  For RPI the CS pin is 0 or 1. you can only use this two. As it referes to SPI inteface control signal 0 or 1 (pins 24-gpio8 & 26-gpio7).
  For the CE pin you can use any gpio you want. In the reference above i used gpio24(pin 18) for CE.

The constructur in the case will be:

 `` var rf24= new nrf24.nRF24(24,0);   
``

As regards pipe number is simple: A pipe number is a 40bit number=that is 5 bytes little endian.
In C++ you can represent this  many ways. I recommend to do this:

Arduino side:
```
byte pipe[5]={ AA, BB, CC , DD , EE , FF };
```
RPI side:
```
var pipe ="0xFFEEDDCCBBAA";
```

Hope this help

Nathanlee2010, 2019-09-19:
thank you very much!   it works!!!

closed 2019-09-20 by ludiazv