ludiazv/node-nrf24 #17: Traing to llisten to slave, or prhebs make some radio scanner

Maksss2018, 2019-10-20:
Hi,
tying to get new  skills in IoT
**here code for master:** 
`const nrf24=require("nrf24"); // Load de module

// Init the radio
var rf24= new nrf24.nRF24(37,24);
rf24.begin();
// Configure the radio
rf24.config({
    PALevel: nrf24.RF24_PA_LOW,
    DataRate: nrf24.RF24_1MBPS,
});

// Register Reading pipes
var pipe=rf24.addReadPipe("0x65646f4e31",true) // listen in pipe "0x65646f4e31" with AutoACK enabled.

// Register callback for reading
rf24.read( function(data,n) {
    // when data arrive on any registered pipe this function is called
    // data -> JS array of objects with the follwing props:
    //     [{pipe: pipe id, data: nodejs with the data },{ ... }, ...]
    // n -> number elements of data array
    console.log(" reciver Onn");
    console.dir(data);

},function(isStopped,by_user,error_count) {
   console.log(" reciver stopped");
   console.log(" isStopped ", JSON.stringify(isStopped));
   console.log(" by_user ", JSON.stringify(by_user));
   console.log(" error_count ", JSON.stringify(error_count));

   // This will be if the listening process is stopped.
});

//rf24.stopRead();

// If you want to abort pending async writes
//rf24.stopWrite();

// Finally to assure that object is destroyed
// and memory freed destroy must be called.
//rf24.destroy();
 
`
**result:**
getting error 101
reciver stopped
 isStopped  true
 by_user  false
 error_count  101

**code for slave:** 
`SPI1.setup({sck:A5, miso:A6, mosi:A7});
var nrf = require("NRF24L01P").connect( SPI1, B0, B1 );
function onInit() {
	nrf.init([0,0,0,0,1], [0,0,0,0,2]);
}
onInit();
setInterval(function() {
	digitalWrite(LED1,1);

	//var value = analogRead(A0); // read a value in
	nrf.sendString("!!!!!!.");
}, 2000);
`

Maksss2018, 2019-10-21:
well, ludiazv, I'm using orange pi pc - I was using this instruction  to connect nrf24l01 + adapter ofr  receiver 
[image]

ludiazv, 2019-10-22:
For Orange Pi you need to have spidev activated in your case spi0 is /dev/spidev0.0. Check constructor documentation CS pin in this case should be 0. (CS=<a>*10+<b> for /dev/spidev<a>.<b>) for the CE pin should be 25 if this is the pin. check the documentation of the orange pi to map the pin to the number in linux.

Maksss2018, 2019-10-22:
Sounds nice - I'll investigate in this way. Thank  you very much

Maksss2018, 2019-10-22:
Well, I've  changed  boot  environment with  `armbian-config` to  . And than  I found that  al this  time  there was no  power 8))))  now  I  wondering  ho  to check is other pinc is  working   becuase  after  I run `ls -l /dev/*spi*`  ...  command return  nothing

`

ludiazv, 2019-10-22:
Without spidev the library can't work.  Check armbian documentation on how to enable spidev overlays for your board. Armbian forums are great to solve this kind of issues. Sorry I can't help with that.

Maksss2018, 2019-10-23:
Thank you  anywhere ...  I've turn on spidev overlay with `armbian-config`  but  steel  no result. I was hoping to so  some-sing with `gpio readall` - no changes  there ..  well  thank  you again, i will try luck on armbian dist forums