ludiazv/node-nrf24 #18: Does not work with node 12

pavelnikitiuk, 2019-11-20:
does not compile with node 12 due to the use of outdated libraries, see details here https://github.com/ludiazv/node-red-contrib-nrf24/issues/4

jojos38, 2020-04-15:
Did you find any fix by the time please?

ludiazv, 2020-04-15:
Yes, we have found the problem but not implemented the solution yet. I had no time do it in the past months. I will try  by next week.

ludiazv, 2020-04-21:
Good news, 
  I've found some time to fix this. I removed the dependency of nan-marshal al copied @pavelnikitiuk  version with minor modifications to the project tree. As the original author of marshal did not update the module in npm.

 This with some minor fixes in the code enable the library to be used with node 8, 10 , 12. Im working in a travis build to automate the check against this versions for arm32 and arm64 platforms. So it will be easy to spot failures with future node versions. 

Changes are nor in the master branch, I will publish a new version in npm when travis build is ready.

Thanks for your contribution.

jojos38, 2020-04-21:
Thank you very much! I can't wait to have the npm version, for now I believe I can download this repo and use npm install [folder of this repo I downloaded]?

ludiazv, 2020-04-21:
I will publish the new version soon in npm so it can be installed via usual npm install

jojos38, 2020-04-21:
Okay thank you!

ludiazv, 2020-04-24:
Released 0.1.5 on npmjs. Now it can be installed via npm install method. 
I checked it with node 10, 12, 13

closed 2020-04-24 by ludiazv