ludiazv/node-nrf24 #20: [Question] installation of NRF25 libs

Trekels, 2020-04-19:
Currently the readmy specifies that the NRF24 libs should be installed, but i noticed that there is an install script in this lib that currently reinstalls them because the check is disabled.

Should this be reenabled or removed from the readme to ensure correcte versions of the lib?

https://github.com/ludiazv/node-nrf24/blob/master/build_rf24libs.sh

ludiazv, 2020-04-20:
Yes
 This is old code form a previous version. The library now requires to use a library with FAILURE HANDLING activated. Therefore we can not trust any version installed already as it could be be and old version or not having this feature.

In the next version I will try to link the library statically into the node module so the system remains untouched or the user can use the dynamic version for CPP code.

Trekels, 2020-04-21:
Just one other note is that the install script configures the NRF24 libs to use the  `--driver=SPIDEV`. This somehow breaks the support for raspberypi.

ludiazv, 2020-04-21:
RPI supports SPIDEV and do not require root you only need to activate it. Let's say that SPIDEV is the "universal" driver for all linux based single boards computers. I do all my testing on a RPI 3.

Trekels, 2020-04-21:
Thats what I thought as well (I am kind of new to this stuff). But I was testing 2 programs.

A very simple one (from the examples folder) where my Rpi(4) listens for incomming data from an arduino(every). Here the RF module `isPresent` works and data is received. In the case of a manual installation of the libs, aswell as installation via the install script.

But if i use a simple mesh network script (also from the example) in the case of a manual install with failure handling enabled and driver RPI, it works. But with the install script it fails (SPIDEV):

`symbol lookup error: /usr/local/lib/librf24network.so.1: undefined symbol: __millis`

This error seems totally unrelated to the `fix` for it but for some reason i cant find a different way to get it running with the install script from the repo.

ludiazv, 2020-04-21:
That's odd. I need to check that. How ever the next version (now published on master) will compile statically the SPIDEV version and will not depend on the version installed in the system.

Trekels, 2020-04-21:
Tested master but no luck. Now setting up the SPI seems to fail`gpio mmap failed: Cannot allocate memory`. When googled suggested fixes are compiling using SPIDEV but it is the oposit in my case. I tested it compiling for BCM2835. and this seems to work just fine.

I also went trought the changes in the NRF24 lib since there is a more recent version available (1.3.4) and i found this [commit](https://github.com/nRF24/RF24/commit/e221acdc4315a39016b31c28143639b27421d75a) which seems to introduce some fixes that might be the reason only the BCM2835 driver works on RPI4

ludiazv, 2020-04-21:
yes. 1.3.4 solves issues with RPI4 driver but breaks SPIDEV. There is a pending change in the lib to change to solve this. We have to wait to the 1.3.5 to try.

I need to test with RPI4 but now I have only one board in use with other project. In principle SPIDEV should work with RPI4. but I have not confirmed it.

ludiazv, 2020-09-25:
The node module has bee updated to new versions of the base libraries. This version should support well Rpi4 now.

The new version of the package will be published in npm