ludiazv/node-nrf24 #21: Most settings not being displayed correctly

jojos38, 2020-04-21:
PA Power    = PA_MIN
IRQ (negative is unused):-1
PayloadSize:32

That's not the full list but a lot of settings are not showed their real values, tho the values I used are indeed applied

ludiazv, 2020-04-21:
could you please post your example and output?

jojos38, 2020-04-21:
Here is a video
https://www.youtube.com/watch?v=YZ1onV1Fmj4

ludiazv, 2020-04-21:
you are showing the radio configuration in begin.
by radio.begin(true). This is used for debug that the radio is connected and always will show default configuration of the radio. change this to radio.begin(false)

If you want to show the radio parameters after config. call radio.config({..},true)
this will show the radio configuration after config.

check it out

jojos38, 2020-04-21:
Oh okay! Thank you!

closed 2020-04-21 by jojos38