ludiazv/node-nrf24 #22: NPM Install fails

jackjury, 2020-05-06:
Hello, 

I have just tried to install this on a raspberry pi, and the install fails, log file below... any ideas?

I have tried running the install command as sudo

`Last login: Tue May  5 15:54:16 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
Jacks-MacBook-Pro-2:~ jackjury$ ssh pi@raspberrypi1.local
Warning: the ECDSA host key for 'raspberrypi1.local' differs from the key for the IP address '192.168.0.42'
Offending key for IP in /Users/jackjury/.ssh/known_hosts:9
Matching host key in /Users/jackjury/.ssh/known_hosts:12
Are you sure you want to continue connecting (yes/no)? no
Host key verification failed.
Jacks-MacBook-Pro-2:~ jackjury$ npm

Usage: npm <command>

where <command> is one of:
    access, adduser, audit, bin, bugs, c, cache, ci, cit,
    completion, config, create, ddp, dedupe, deprecate,
    dist-tag, docs, doctor, edit, explore, get, help,
    help-search, hook, i, init, install, install-test, it, link,
    list, ln, login, logout, ls, outdated, owner, pack, ping,
    prefix, profile, prune, publish, rb, rebuild, repo, restart,
    root, run, run-script, s, se, search, set, shrinkwrap, star,
    stars, start, stop, t, team, test, token, tst, un,
    uninstall, unpublish, unstar, up, update, v, version, view,
    whoami

npm <command> -h  quick help on <command>
npm -l            display full usage info
npm help <term>   search for help on <term>
npm help npm      involved overview

Specify configs in the ini-formatted file:
    /Users/jackjury/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config

npm@6.4.1 /usr/local/lib/node_modules/npm
Jacks-MacBook-Pro-2:~ jackjury$ ssh pi@raspberrypi1.local
Warning: the ECDSA host key for 'raspberrypi1.local' differs from the key for the IP address '192.168.0.42'
Offending key for IP in /Users/jackjury/.ssh/known_hosts:9
Matching host key in /Users/jackjury/.ssh/known_hosts:12
Are you sure you want to continue connecting (yes/no)? yes
pi@raspberrypi1.local's password:
Linux raspberrypi1 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed May  6 13:14:17 2020 from 192.168.0.15
pi@raspberrypi1:~ $ mkdir v4node
pi@raspberrypi1:~ $ cd v4node
pi@raspberrypi1:~/v4node $ npm install
-bash: npm: command not found
pi@raspberrypi1:~/v4node $ sudo apt-get npm
E: Invalid operation npm
pi@raspberrypi1:~/v4node $ node
-bash: node: command not found
pi@raspberrypi1:~/v4node $ uname -m
armv6l
pi@raspberrypi1:~/v4node $ wget https://nodejs.org/dist/v8.9.0/node-v8.9.0-linux-armv6l.tar.gz
[... 433 lines cut ...]
4 verbose npm-session 8913e67e3ea60600
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/nrf24 2767ms (from cache)
8 silly pacote tag manifest for nrf24@latest fetched in 3320ms
9 silly install loadIdealTree
10 silly install cloneCurrentTreeToIdealTree
11 silly install loadShrinkwrap
12 silly install loadAllDepsIntoIdealTree
13 silly resolveWithNewModule nrf24@0.1.5-beta checking installable status
14 http fetch GET 304 https://registry.npmjs.org/bindings 1243ms (from cache)
15 silly pacote range manifest for bindings@^1.5.0 fetched in 1324ms
16 silly resolveWithNewModule bindings@1.5.0 checking installable status
17 http fetch GET 304 https://registry.npmjs.org/nan-check 1310ms (from cache)
18 http fetch GET 304 https://registry.npmjs.org/nan 1343ms (from cache)
19 silly pacote version manifest for nan-check@0.0.6 fetched in 1407ms
20 silly resolveWithNewModule nan-check@0.0.6 checking installable status
21 silly pacote range manifest for nan@^2.14.0 fetched in 1447ms
22 silly resolveWithNewModule nan@2.14.1 checking installable status
23 http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 575ms (from cache)
24 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 648ms
25 silly resolveWithNewModule file-uri-to-path@1.0.0 checking installable status
                                                                [ Read 200 lines ]
^G Get Help     ^O Write Out    ^W Where Is     ^K Cut Text     ^J Justify      ^C Cur Pos      M-U Undo        M-A Mark Text   M-] To Bracket
^X Exit         ^R Read File    ^\ Replace      ^U Uncut Text   ^T To Spell     ^_ Go To Line   M-E Redo        M-6 Copy Text   ^Q Where Was`

jackjury, 2020-05-06:
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node',
1 verbose cli   '/usr/local/bin/npm',
1 verbose cli   'install',
1 verbose cli   'nrf24',
1 verbose cli   '--save' ]
2 info using npm@5.5.1
3 info using node@v8.9.0
4 verbose npm-session 8913e67e3ea60600
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/nrf24 2767ms (from cache)
8 silly pacote tag manifest for nrf24@latest fetched in 3320ms
9 silly install loadIdealTree
10 silly install cloneCurrentTreeToIdealTree
11 silly install loadShrinkwrap
12 silly install loadAllDepsIntoIdealTree
13 silly resolveWithNewModule nrf24@0.1.5-beta checking installable status
14 http fetch GET 304 https://registry.npmjs.org/bindings 1243ms (from cache)
15 silly pacote range manifest for bindings@^1.5.0 fetched in 1324ms
16 silly resolveWithNewModule bindings@1.5.0 checking installable status
17 http fetch GET 304 https://registry.npmjs.org/nan-check 1310ms (from cache)
18 http fetch GET 304 https://registry.npmjs.org/nan 1343ms (from cache)
19 silly pacote version manifest for nan-check@0.0.6 fetched in 1407ms
20 silly resolveWithNewModule nan-check@0.0.6 checking installable status
21 silly pacote range manifest for nan@^2.14.0 fetched in 1447ms
22 silly resolveWithNewModule nan@2.14.1 checking installable status
23 http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 575ms (from cache)
24 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 648ms
25 silly resolveWithNewModule file-uri-to-path@1.0.0 checking installable status
26 http fetch GET 304 https://registry.npmjs.org/nan-marshal 296ms (from cache)
27 silly pacote version manifest for nan-marshal@0.0.5 fetched in 334ms
28 silly resolveWithNewModule nan-marshal@0.0.5 checking installable status
29 silly currentTree tempmonnode@1.0.0
29 silly currentTree ├── bindings@1.2.1
29 silly currentTree ├── epoll@0.1.22
29 silly currentTree ├── fifolock@1.0.0
29 silly currentTree ├── nan@2.6.2
29 silly currentTree ├── nrf@0.8.3
29 silly currentTree ├── pi-pins@1.0.2
29 silly currentTree ├─┬ pi-spi@1.2.1
29 silly currentTree │ └── nan@2.14.1
29 silly currentTree └── queue-async@1.0.7
30 silly idealTree tempmonnode@1.0.0
30 silly idealTree ├── bindings@1.2.1
30 silly idealTree ├── epoll@0.1.22
30 silly idealTree ├── fifolock@1.0.0
30 silly idealTree ├── file-uri-to-path@1.0.0
30 silly idealTree ├── nan-check@0.0.6
30 silly idealTree ├── nan-marshal@0.0.5
30 silly idealTree ├── nan@2.6.2
30 silly idealTree ├── nrf@0.8.3
30 silly idealTree ├─┬ nrf24@0.1.5-beta
30 silly idealTree │ ├── bindings@1.5.0
30 silly idealTree │ └── nan@2.14.1
30 silly idealTree ├── pi-pins@1.0.2
30 silly idealTree ├─┬ pi-spi@1.2.1
30 silly idealTree │ └── nan@2.14.1
30 silly idealTree └── queue-async@1.0.7
31 silly install generateActionsToTake
32 silly diffTrees action count 6
33 silly diffTrees add file-uri-to-path@1.0.0
34 silly diffTrees add nan-marshal@0.0.5
35 silly diffTrees add nan-check@0.0.6
36 silly diffTrees add bindings@1.5.0
37 silly diffTrees add nan@2.14.1
38 silly diffTrees add nrf24@0.1.5-beta
39 silly decomposeActions action count 48
40 silly decomposeActions fetch file-uri-to-path@1.0.0
41 silly decomposeActions extract file-uri-to-path@1.0.0
[... 106 lines cut ...]
148 warn tempmonnode@1.0.0 No repository field.
149 verbose stack Error: nrf24@0.1.5-beta preinstall: `./build_rf24libs.sh`
149 verbose stack Exit status 1
149 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:280:16)
149 verbose stack     at emitTwo (events.js:126:13)
149 verbose stack     at EventEmitter.emit (events.js:214:7)
149 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
149 verbose stack     at emitTwo (events.js:126:13)
149 verbose stack     at ChildProcess.emit (events.js:214:7)
149 verbose stack     at maybeClose (internal/child_process.js:925:16)
149 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
150 verbose pkgid nrf24@0.1.5-beta
151 verbose cwd /home/pi/v4node
152 verbose Linux 4.19.97+
153 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "nrf24" "--save"
154 verbose node v8.9.0
155 verbose npm  v5.5.1
156 error code ELIFECYCLE
157 error errno 1
158 error nrf24@0.1.5-beta preinstall: `./build_rf24libs.sh`
158 error Exit status 1
159 error Failed at the nrf24@0.1.5-beta preinstall script.
159 error This is probably not a problem with npm. There is likely additional logging output above.
160 verbose exit [ 1, true ]
`

jackjury, 2020-05-06:
[fail.log](https://github.com/ludiazv/node-nrf24/files/4587785/fail.log)

ludiazv, 2020-05-06:
Hi did you tried with node >=10?

jackjury, 2020-05-07:
Yes I have now thank you, its still not 100% working, but I'll close this issue!

closed 2020-05-07 by jackjury