ludiazv/node-nrf24 #23: Hardware Not Found

jackjury, 2020-05-07:
I have been trying to get 1 x Arduino Uno and 1 x Raspberry Pi 1 (Model B) talking for a couple of weeks now.  

I am sure this isn't a hardware issue, as I can get the gettingstarted example in rf24libs working, but I can't make any of the node versions to work.  

Each time node seems to fail to detect the hardware, despite me configuring the pins correctly.

I have had similar issues with python ports too, and any of the other lib examples.  Any ideas on what might be causing this?

jackjury, 2020-05-07:
I am running node v10.20.1

ludiazv, 2020-05-07:
Hi,
   I have not tested with RPI1. But in principle should work. Can you provide some additional information

- Sample code and wiring you are using? For troubleshooting I recommend you to use the Gettingstarted.js in the examples folder modify the config of the radio with your wiring. This sript works with getting started of arduino.
- Result of the output of your code.
- OS version and SPIDEV configuration?