ludiazv/node-nrf24 #24: Advanced features

ahmetertem, 2020-10-15:
First of all, thanks to those who collaborated on this project. 

I want to know when dynamic payloads and acknowledge payload will be ready.

---

Right now I'm working on a project which communicates between RPI and Arduino. Somehow the packages which sent from Arduino can't auto ack. (*Arduino-to-Arduino works fine*). So that's why I thought it's because I'm using dyn.payload and autoack at Arduino, but this library is not supporting.

Note: I enabled auto_ack at openreadpipe (*and I can see it in info*). I'm receiving the package at `nrf24.read` and it matches exactly but Arduino is not understanding acknowledge.

Thanks again

ludiazv, 2020-10-16:
Hi,
    This to advances features are not supported yet. At some point they should be implemented but I do not know when, specially as testing requires time for this embedded development. PRs are welcomed. 

On they other hand, AFAIK both radios must be configured with dyn payloads to communicate properly so this is the most probably the reason why the Arduino is not getting the ACK back.

ahmetertem, 2020-10-16:
Hi @ludiazv,
Thanks for the reply.
I'll test and inform ASAP

Regards

ahmetertem, 2020-10-17:
@ludiazv ; 

As you said when both payload sizes are same and dynamic payload is disabled it's working.

Thank you very much

closed 2020-10-17 by ahmetertem