ludiazv/node-nrf24 #25: Reading strings and converting C++ structs

stoopfrench, 2020-10-19:
Do you have any insight into how to use this library to read strings sent from a radio on an arduino?  It's simple enough to read an integer sent from the arduino radio but I cant figure out how to convert the buffer in node back into a string - I've tried node Buffers .toString() method but nothing gets returned. 

I was also wondering if you have any idea how to send something from this library that could be used as a struct on the arduino side?  I've passed structs between different radios running on arduinos but I cant figure out how to create a buffer in node that could be used as a struct when sent to an arduino. 

So far the library is great! Thanks for any help.

ludiazv, 2020-10-19:
Hi @stoopfrench 
  The library works at link level. It only enable to send/receive frames. The frame payload is a binary blob. How to treat it is application specific. Converting a C null-terminated string in ascii is easy, iterate over the received buffer until you find '\0' and don't forget to send the null char from the Arduino.

  I do not recommend to send C/C++ structs directly over the radio. There are many factors that affect the result: mcu/compiler endianess, struct alignment, struct packaging. This depend on the compiler options and pragmas used in the Arduino. There are ,however, npm modules that enable you to interpret the buffer with a struct semantic like python pack/unpack.

I prefer to define the format of the frame and construct the frame explicitly. And then decode it explicitly on the other side. I use this library for sensors and I use standard CayenneLPP format. There are nodejs decoders/encoders and arduino libraries to encode/decode this format.

stoopfrench, 2020-10-20:
@ludiazv 

Thanks for the info! CayenneLPP was the key to solving my problems - i had never heard of it. Again, thanks for the library!