ludiazv/node-nrf24 #26: build with RPi driver : radio not working or in use

Marsu31, 2020-12-28:
Hi !

I installed this library with the command line : `DRIVER=RPi npm install nrf24@0.1.7-beta --save`
When I use your nRF24input node I get these error messages :

- flow :  _RF24 radio not working or in use_
- log : _[info] [RF24radio:module radio] Radio Failed to initialize, wiring or hardware problem. FailureDetected:true_

The same flow works fine when I use the default driver (SPIDEV).
Enabling or disabling SPI (with raspi-config) doesn't change anything.
Using the getting_started.py from https://tmrh20.github.io/RF24/ works.

Regards

ludiazv, 2020-12-28:
why do you need to use the Rpi driver?

Marsu31, 2020-12-28:
I want to connect the nRF24 module to my RPi 4 which OS is Debian (not Raspbian). Debian for RPi 4 doesn't yet support SPI devices.
Before going in "production" (at home ! ;-) ), for testing, I use a RPi 3 with Raspbian.

ludiazv, 2020-12-28:
I see, I have not tested this vanilla Debian. I'm using a RPI3B for testing and the RPi driver works.  But it has some caveats:

- Run node as root (this is a security problem specially if you plan to run nodered)
- Disable spi=on on raspiOS to avoid load the spidev driver.
- Reboot the system.

In order to test you can use the Gettingstarted.js in examples. Change the CE const in the line with your pin:

```bash
$ mkdir tst
$ cd tst
$ DRIVER=RPi npm install nrf24@0.1.7-beta --save
$ cd node_modules/nrf24
$ sudo node examples/GettingStarted.js
```

you should see something like this:
````
...
===========================
Radio details after config:
===========================
================ SPI Configuration ================
CSN Pin  	 = 0
CE Pin  	 = Custom GPIO24
SPI Speedz	 = 10 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x00
CONFIG		 = 0x7e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MIN
Config internals:
check the values as incorrect values fallback to default values.
 PALevel:0 EnableLna:0 DataRate:0 Channel:76 CRCLength:2 Retries:(5+1*250us,15 attempts) PayloadSize:32 AddressWidth:5 AutoAck:1 TxDelay:250 PollBaseTime:35000 IRQ (negative is unused):-1 AutoFailureRecovery:0
===========================
RF24 GettingStarted
This script assumes that this radio is radionumber=1 in ardunino.
Other radio must be configured with radionumber=0
[s]ender or [R]eceiver?
```

Marsu31, 2020-12-28:
Thank you.
I think there is some relation with the chapter **Running as root** from the [BCM2835 documentation](https://www.airspayce.com/mikem/bcm2835/). I added the argument `--cpu-flags="-DBCM2835_HAVE_LIBCAP"` to the configure command in your `build_rf24libs.sh` file, line 25. The build worked. I also did all the configuration steps but I got an error about lib cap when I start node-red (as pi user) : _undefined symbol: "cap_get_proc"_.
I think the problem is not from your code. I'm sorry to have bothered you.

Regards

ludiazv, 2020-12-28:
Possibly you need to link the libcap library to manage linux capabilities. try -lcap in the bindings.gyp

This however is a hacky way of running as not root. I think is better to use SPIDEV and mange the persmissions at fs level.

I don't know but maybe is possible to load the spidev driver in debian.

Regards,

Marsu31, 2020-12-28:
Well done ! The cap error has disappeared but I got this now : _ERROR:Radio is not ready!_
It works with sudo.
I think you are right. I should manage to use spidev. Debian can't but Ubuntu does... ;-)

ludiazv, 2020-12-29:
Did you try diet-pi? I think think is a mature option to run raspiOS without all the bloat of regular raspbian.

closed 2020-12-29 by ludiazv

Marsu31, 2020-12-29:
Hi !
Nice idea. I didn't know this OS. 64bits for RPi is still beta. I need it
to run docker. I'll try it on an other device. Thank you. RockPro64 seems
to be a good sbc for my needs...
Regards

ludiazv, 2020-12-30:
RockPro64 is a powerful board indeed. Armbian supports it so in principle you can run a stable 64bit OS with docker on this board. 
I can confirm that this module runs well in armbian arm64 using spidev and node-red in a opi zero plus. I've never used the library inside a cointainer but in it should work as long as the container has access to the spi /dev and the /sys/.../gpio fs (using docker --device , --volume & --privileged flags)

To avoid power and wiring problems I built [this](https://github.com/ludiazv/borosRF2) carrier board that works nice with regular nrf24l01 modules including those with PALNA for better range.

Happy hacking!

Marsu31, 2020-12-30:
I will use my old pine64 board :
[image: 20201230_194319_resize.jpg]
It doesn't work very well. There are many loss.
I don' t know if the IRQ pin would improve it, not how to use it... I have
a lot of work but I will succeed ;-)

See you ;-)

ludiazv, 2020-12-31:
In my experience packet loss is more related to hardware issues than  software. (power, decoupling, wiring, antena ...). Software parameters have an impact: Less data rate and higher PA Level will reduce packet loss.IRQ helps if the number of packets per second is high. For example if you transfer only one packet per second IRQ line make no difference.  The other advantage of IRQ is to reduce CPU usage.

Marsu31, 2020-12-31:
I don't know why there were packet loss. It was only with JS code. The
python worked great.
I re-installed with DietPi like you suggested ;-)  Thank you. I like to
learn...
User rights are not trivial. But now, it works :-) By default, GPIOs are
accessible only for root.
My new problem is correction. I moved my Pine64 in the garage and the
signal is low, even with antenna (sender and receiver), max power and IRQ.
I search a clean solution to guarantee communications (I think that ACKs
are not always received because the emitter repeat its messages). I send
messages which payload size is 27 bytes and it will occurs every 15
minutes. I don't need Mbits :-D
LoraWAN would be a better solution for my needs but it is too expensive :'(