ludiazv/node-nrf24 #29: GettingStarted.js cannot read more than 8 bytes on the buffer

Radio-Op, 2021-02-08:
Thank you for your outstanding work.

I have a great interest for radio and networking, unfortunatly I am not a dev, so my comments might be not smart at all...
Anyway..

I use this to readthe buffer and print the data.
An arduino is sending two values (from two sensors) grouped in a structure (weather sensors)
---
radio.read(function(d,items) {
   for(var i=0;i<items;i++){
      let r=d[i].data.readFloatLE(0);
      console.log("Payload Rcv [" + r +"], reply result:" + radio.write(d[i].data));
      console.log(d);
---

the results are : 

{
    pipe: 1,
    data: <Buffer 4a 36 48 3f 00 00 ce 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
  }
]
Payload Rcv [0.7820783853530884], reply result:1

As yuo can see, the first  4 bytes " 4a 36 48 3f " are decoded as  0.78207...  This is a correct value from the first sensor (Checked on the console of the arduino)
However the following 4 bytes "00 00 ce 42" ( 103 ) are not decoded

I would like the two 4 bytes to be decoded and put in a structure too in order to be forwarded to some websocket process.

Thanks you By advance.
Regards.

Vincent.

ludiazv, 2021-02-09:
Hi @Radio-Op ,
  You need to decode the numbers your self as the buffer is a binary blob of fixed size. In play nodejs you do it with ```readFloatLE(offset)```. In your case you should:

```javascript
...
   let sensor1=d[i].readFloatLE(0);  // decode first 4-byte float in little endian starting at the beginning of the buffer.
   let sensor2=d[i].readFloatLE(4); // decode second float starting in the 5th byte of the buffer
...

```
 There are nodejs modules to deal with packed structures inside a the binary buffer.  Check for example this thread https://stackoverflow.com/questions/5605108/pack-unpack-functions-for-node-js for python way of packing/unpacking or 
this module https://www.npmjs.com/package/struct for a declarative C-like implementation.

Regards,

Radio-Op, 2021-02-09:
Thanks a lot.
You can if you wish remove my question as it is not related to the library
itself.
After studying the code (I am not familiar with coding) I 've found a
command to copy a buffer into another one that could also work.
:)

Again thank you.

Vincent

closed 2021-02-09 by ludiazv