ludiazv/node-nrf24 #30: ASUS Tinker board CE

gedeondt, 2021-04-17:
Hello,

I can not manage to get it working fine on a tinker boar.

I spent a lot of hours trying and, by change, I could send some data when I disconnected CE completely.

The problem seems to be related to GPIO.

var rf24= new nrf24.nRF24(22, 20);

or

var rf24= new nrf24.nRF24(15, 20);

The CE pin is wired to the pin GPIO5B7 (tinker), physical pin 15, GPIO22. If I keep it wired:

RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.

If I disconnect the wire it works.

Any idea?

ludiazv, 2021-04-19:
Hola @gedeondt,
   No tengo esta placa así que no te lo puedo confirmar. Pero me parece que el problema es cómo se asignan en esta placa los numeros de los GPIOS. Según este hilo: https://tinkerboarding.co.uk/forum/archive/index.php/thread-306.html la formula que debes usar es esta:

````
GPIO marking consists of GPIO[0-8][A-D][0-7] eg. 9x 32 bit GPIO banks are split to 4 ("A-D") 8 bit ("0-7") pins (total 160 pins some are unused/unrouted). Example of translation to number:
GPIO0C1 = 0*32 + ("C") 2*8 + 1 = 17
GPIO5B4 = 5*32 + ("B") 1*8 + 4 - 8 = 164 
````
En tu caso si usas el GPIO5B7 = 5*32 + 8 + 7- 8 = 174.

El parámetro CE debe ser **174**

Saludos,

gedeondt, 2021-04-22:
Muchas gracias @ludiazv

Efectivamente :)

La verdad es que es una buena placa pero falta una documentación más fina de ciertas cosas.

Gracias!