ludiazv/node-nrf24 #31: Error writing to pipe depending on scope

gedeondt, 2021-04-26:
Hi,

   I am having an issue regarding this code:

          var nrf24=require("nrf24");
          var readline = require('linebyline');
          var rf24= new nrf24.nRF24(174, 20);
          console.log("begin->",rf24.begin(true));
          rf24.useWritePipe("0x72646f4e31");
          var rl = readline(process.stdin);
          
          rl.on('line', function(line){
                  success = rf24.write(Buffer.from(line));  ------------> This write works
                  console.log("Sent " + ( success ? "OK" : "KO" ));
          });
          
          success = rf24.write("ASD");    ---> This fails: SyntaxError: Unknown error

      I can't see the difference between both. It's weird.

ludiazv, 2021-04-26:
Hi,
   ```write``` requires a node Buffer as parameter. The SyntaxError is produced by the library as you passed a string in the second call. ```success = rf24.write(Buffer.from("ASD"));``` should work.

gedeondt, 2021-04-27:
Hi,

    Thanks! My fault.

    Closed.

closed 2021-04-27 by gedeondt