ludiazv/node-nrf24 #35: Installatio falilure

ghostmaster75, 2022-03-25:
i am trying to install the package on raspbian 11 bullseye 64bit - nodejs v14.19.1 but i have this error:

```
g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
g++: error: unrecognized command-line option ‘-marm’
g++: error: unrecognized command-line option ‘-mfpu=vfp’
g++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make: *** [Makefile:42: RF24.o] Errore 1
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! nrf24@0.1.7-beta preinstall: `./build_rf24libs.sh`
npm ERR! Exit status 2
npm ERR!
npm ERR! Failed at the nrf24@0.1.7-beta preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-03-25T14_52_04_566Z-debug.log
```

this is the log file

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install', 'nrf24' ]
2 info using npm@6.14.16
3 info using node@v14.19.1
4 verbose npm-session 3fdf41ab69052fba
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/nrf24 1496ms (from cache)
8 silly pacote tag manifest for nrf24@latest fetched in 1594ms
9 timing stage:loadCurrentTree Completed in 2331ms
10 silly install loadIdealTree
11 silly install cloneCurrentTreeToIdealTree
12 timing stage:loadIdealTree:cloneCurrentTree Completed in 5ms
13 silly install loadShrinkwrap
14 timing stage:loadIdealTree:loadShrinkwrap Completed in 122ms
15 silly install loadAllDepsIntoIdealTree
16 silly resolveWithNewModule nrf24@0.1.7-beta checking installable status
17 http fetch GET 304 https://registry.npmjs.org/bindings 594ms (from cache)
18 silly pacote range manifest for bindings@^1.5.0 fetched in 642ms
19 silly resolveWithNewModule bindings@1.5.0 checking installable status
20 http fetch GET 304 https://registry.npmjs.org/nan 633ms (from cache)
21 silly pacote range manifest for nan@^2.14.0 fetched in 666ms
22 silly resolveWithNewModule nan@2.15.0 checking installable status
23 http fetch GET 304 https://registry.npmjs.org/nan-check 1093ms (from cache)
24 silly pacote version manifest for nan-check@0.0.6 fetched in 1122ms
25 silly resolveWithNewModule nan-check@0.0.6 checking installable status
26 http fetch GET 304 https://registry.npmjs.org/file-uri-to-path 191ms (from cache)
27 silly pacote version manifest for file-uri-to-path@1.0.0 fetched in 208ms
28 silly resolveWithNewModule file-uri-to-path@1.0.0 checking installable status
29 http fetch GET 304 https://registry.npmjs.org/nan-marshal 808ms (from cache)
30 silly pacote version manifest for nan-marshal@0.0.5 fetched in 827ms
31 silly resolveWithNewModule nan-marshal@0.0.5 checking installable status
32 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 2406ms
33 timing stage:loadIdealTree Completed in 2581ms
34 silly currentTree pi
34 silly currentTree ├── node-red-contrib-rpi-433@1.3.0
34 silly currentTree ├── q@1.5.1
34 silly currentTree ├── rpi-433-v3@1.2.0
34 silly currentTree └── underscore@1.13.2
35 silly idealTree pi
35 silly idealTree ├── bindings@1.5.0
[... 155 lines cut ...]
167 warn pi No repository field.
168 warn pi No README data
169 warn pi No license field.
170 verbose stack Error: nrf24@0.1.7-beta preinstall: `./build_rf24libs.sh`
170 verbose stack Exit status 2
170 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
170 verbose stack     at EventEmitter.emit (events.js:400:28)
170 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
170 verbose stack     at ChildProcess.emit (events.js:400:28)
170 verbose stack     at maybeClose (internal/child_process.js:1058:16)
170 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:293:5)
171 verbose pkgid nrf24@0.1.7-beta
172 verbose cwd /home/pi
173 verbose Linux 5.10.103-v8+
174 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "nrf24"
175 verbose node v14.19.1
176 verbose npm  v6.14.16
177 error code ELIFECYCLE
178 error errno 2
179 error nrf24@0.1.7-beta preinstall: `./build_rf24libs.sh`
179 error Exit status 2
180 error Failed at the nrf24@0.1.7-beta preinstall script.
180 error This is probably not a problem with npm. There is likely additional logging output above.
181 verbose exit [ 2, true ]
```

ludiazv, 2022-03-27:
Hi,
 The module uses a c++ library that have some problems on 64bit Raspi OS versions. I will look if there is any update now on this.

ludiazv, 2022-03-29:
Updated to 0.2 version that solves the issue. It includes prebuilt binaries for node 12,14,16 & 17 for arm32 and arm64. Installation should be quite fast as the binaries are pre-compiled in the package.

ghostmaster75, 2022-03-29:
Thank you very much!